Qucs/ADMS #21: XML::LibXML parent.pm not found

jymag2, 2015-07-06:
I need to install ADMS from this source, but when trying to install XML-LIBXML-2.0121, after having installed all the prerequisites, I get an error saying that the parent.pm file in many test cases is not found. Do you have any idea how I can remedy this problem? Thank you for you time and help.

guitorri, 2015-07-06:
Which system are you using? Which Perl version?
The error is during the XML::LibXML installation or after when you are trying to build the code?

The XML::LibXML Perl module can be a bit annoying to install...

On Linux, with older Ubuntu, this should work:
https://github.com/Qucs/ADMS/blob/master/.travis.yml#L16-L20

Newer Ubuntu/Debian versions have a package for that `libxml-libxml-perl`.

On Windows you could try with Strawberry perl, which already have XML::LibXML included:
https://github.com/Qucs/ADMS/blob/master/.appveyor.yml

That said. I you install from the released tarball (http://sourceforge.net/projects/mot-adms/files/adms-source/2.3/) , Perl and XML::LibXML should not be necessary at all. It is needed only if you build form the Git repository.

jymag2, 2015-07-06:
Linux, perl 5.10.1. I believe I have figured it out, I need to install the
parent Perl module. However, I have installed ADMS from the tarball on
sourceforge, but I thought it was outdated. Is it the same as the version
on github? If so, thank you very much for letting me know. Thank you for
your time and help once again.

jymag2, 2015-07-06:
I am running Perl 5.10.1, Linux 2.6.32-504.16.2.el6.x86_64.
I successfully installed the XML::LIBXML. When I tried installing ADMS from
the github zip file, after sh bootstrap.sh, the following erros were
listed, among others: autom4te: /usr/bin/m4 failed with exit status: 1,
aclocal: autom4te failed with exit status: 1, no proper invocation of
AM_INIT_AUTOMAKE was found, MAINTAINER_MODE does not appear in
AM_CONDITIONAL, am__fastdepCC does not appear in AM_CONDITIONAL,
configure.ac:30: error: possibly undefined macro: AM_MAINTAINER_MODE,
configure.ac:91: error: possibly undefined macro: AC_LIBTOOL_WIN32_DLL.
When I ran make, it says: make: *** No targets specified and no makefile
found.  Stop.

If the ADMS from sourceforge offers the same functionality as the one from
github, that would be great. I am running Xyce 6.2, and the sourceforge
ADMS says it is for Xyce 6.1, which is why I thought it was outdated.

Thank you very much.

guitorri, 2015-07-06:
Is `automake` installed?

The GitHub zip is just a snapshot of the repository (maintainer mode is needed to generate source files). 

The tarball on SourceForge is a pre-configured package (maintainer mode is not necessary). The auto generated sources are included. A `./configure && make install` should work.

Version > 2.3.3 should be good for Xyce 6.2.

closed 2015-07-07 by jymag2