Qucs/ADMS #22: Analog functions undefined

jymag2, 2015-07-07:
I am trying to add verilog-A logic gate files into a simulator. The code I am using implements the transition() function, which ADMS returns as an undefined function, also the always functions is not found. Is there a manual that has the currently supported functions of ADMS? I am using the ADMS from the sourceforge site. Thank you for your time and help.

guitorri, 2015-07-09:
For the `transition()` there is a [fix available](https://github.com/Qucs/ADMS/compare/master...guitorri:transition?expand=1). It should be merged soon.

For the `always` block I do not know. I guess some work is necessary on the parser and the internal tree.

Unfortunately there is no list with the supported constructs.

tvrusso, 2015-07-09:
I should point out, though, that applying the fix to ADMS will not automatically get access to the transition function in any given simulator unless the back-end code generation XML knows how to generate the right code.  You are using Xyce for your simulator, and Xyce's ADMS back-end does not support either the transition digital function or always blocks.  Even if you got ADMS to recognize your input verilog with this fix, you'd be on your own to hack the code generation to work for your model.    There is no plan for the Xyce project to support the digital modeling parts of Verilog-A at the moment.

guitorri, 2015-07-23:
The `transition()` fix is now merged. The `always` will need work and I cannot do it right now.

closed 2015-07-23 by guitorri