Qucs/ADMS #33: XML::LibXml not found

crobarcro, 2015-10-20:
I'm having a go building ADMS using cmake from the git repo on a Mint Linux system (based on Ubuntu 14.04  	Ubuntu Trusty). I'm getting the following cmake error:
```
$ cmake ~/src/qucs-git/qucs-core/deps/adms/ -DUSE_MAINTAINER_MODE=ON -DCMAKE_BUILD_TYPE=RELEASE
-- Checking for perl module XML::LibXml
-- Checking for perl module XML::LibXml - failed
-- Could NOT find PerlModules (missing:  PERLMODULES_XML_LIBXML_MODULE) 
CMake Error at CMakeLists.txt:41 (message):
  Not all required perl modules were found on your system
```
However, I have the `libxml-libxml-perl` module installed, is this the right package? Any idea what might be going wrong here?

guitorri, 2015-10-20:
Perhaps `libxml2 libxml2-dev` are also needed?

Can you run this:
`$ perl -e "use XML::LibXML"`

crobarcro, 2015-10-20:
the perl command gives no output whatsoever, and I definitely have both these packages installed.

guitorri, 2016-10-28:
I hope this issue is solved by now. Please reopen if needed.

closed 2016-10-28 by guitorri