Qucs/ADMS #35: License issue: Verilog-AMS headers provided by Accelera

guitorri, 2016-01-28:
The Verilog-AMS headers provided by Accelera (http://accellera.org/downloads/standards/v-ams) are non-free, hence incompatible with (L)GPL. 

They do not allow for **modification**, only *use* and *distribution*. The latest copyright notice states:

```
// Copyright(c) 2009-2014 Accellera Systems Initiative Inc.
// 1370 Trancas Street #163, Napa, CA 94558, USA.
//
// The material in constants.vams is an essential part of the Accellera Systems
// Initiative ("Accellera") Verilog-AMS Language Standard. Verbatim copies of
// the material in this Annex may be used and distributed without restriction.
// All other uses require permission from Accellera IP Committee
// (ipr-chair@lists.accellera.org).
// All other rights reserved
//
// Version 2.4.0
//
```

ADMS should not have these files in the repository, and ADMS should not be distributed with these files. 

There are few alternatives:
 - Convince Accellera to change their copyright notice to something that is (L)GPL friendly.
 - Implement headers from scratch, see [clean room proposal](https://github.com/Qucs/ADMS/wiki/Header-license-issue#clean-room-implementation).
 - Provide our own non-compliant header and means to fetch the non-free compliant ones from another package ([from this debian-legal suggestion](https://lists.debian.org/debian-legal/2015/01/msg00015.html))

Further details from previous discussions on this subject can be found below:
 - https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=501264
 - https://lists.debian.org/debian-legal/2013/07/msg00042.html
 - https://lists.debian.org/debian-legal/2015/01/msg00011.html
 - http://sourceforge.net/p/mot-adms/patches/4/

felix-salfelder, 2016-01-28:
@guitorri i have computed some numbers and (almost) implemented the header that bastien describes in https://github.com/Qucs/ADMS/wiki/Header-license-issue#clean-room-implementation

can you pick it from there, add `#define` and commit it with a (GPLv2) header?

guitorri, 2016-01-28:
@MikeBrinson thank you for the reminder:
> Hello Felix
Writing a simple, experimental, set of headers should not be thatdifficult.  It could be based on the published data in
K.S. Kundert and Olaf Zinke, "The Designer's Guide to Verilog-AMS",
Kluwer Academic Publishers, Boston/Dordrecht/London, First Edition, June 2004,ISBN: 1-4020-8044-1.
Mike

I thought about this book and was concerned about the book publisher copyright. It guess I was right on being concerned. See below screenshot from Amazon (taken on the grounds of fair use). 

If I read it correctly, to just *execute* the pieces of code in the book one has to purchase a copy of the book? Is that right? Please someone confirm. 

[image]

It seems to be a similar [issue](https://www.gnu.org/software/gsl/design/gsl-design.html) with the Numerical Recipes book (among others). The authors/publishers put restrictions that make the code they provide unusable with GPL.

guitorri, 2016-02-06:
Headers have been replaced.

closed 2016-02-06 by guitorri