Qucs/ADMS #36: ADMS license, LGPLv2 or GPLv2?

guitorri, 2016-01-31:
The `COPYING` file states contains the following header:

```
GNU Lesser General Public License
Version 2.1, February 1999
```

From the above I assumed ADMS was LGPLv2. Fact that I wrote in the `README.md`.

However the file `scripts/vlatommlMODULE.htm.xml` states the following:

```
      This file is part of adms - http://sourceforge.net/projects/mot-adms.
      adms is free software; you can redistribute it and/or modify
      it under the terms of the GNU General Public License as published by
      the Free Software Foundation; either version 2 of the License, or
      (at your option) any later version.
```

This file was last updated by the author in 2006:

```
   $Id: vlatommlMODULE.htm.xml 1156 2008-11-10 15:28:15Z r29173 $
   $Log$
   Revision 1.19  2006/03/09 13:35:02  r29173
```

In the code examples in `scripts/analogfunction.xml` the author decided to generate example code with the following headers:

```
<admst:open file="analogfunction.h">/*
 * analogfunction.h - analog function header
 *
 * This is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2, or (at your option)
 * any later version.
 * 
 */
```

Note that the GPLv2 was published months after the LGPLv2.

```
GNU GENERAL PUBLIC LICENSE
Version 2, June 1991
```

Can we safely assume that the author changed to GPL2+ and forgot to update the COPYING file?

lutherthecat, 2016-01-31:
My understanding is that the GPL is file-based.  It is entirely possible 
that part of the project may be LGPL and the rest is GPL. The nice thing 
about the LGPL is that you are allowed to convert the license to the 
GPL.  In addition, you can go from version 2 to version 3 of the 
license, unless the author changed the license term that it cannot be 
converted.

You would have to look at each file to see whether or not the authors 
have changed:

|either version 2 of the License, or (at your option) any later version. |

to

|version 2 of the License. |

For the pyadms project, I have converted the license from LGPL v2.1 to 
GPL v 2.0.  In addition, I have made it version 2 only.

felix-salfelder, 2016-02-01:
> For the pyadms project, I have converted the license from LGPL v2.1 to GPL v 2.0. In addition, I have made it version 2 only.

thanks, this seems to make sense. then we should change COPYING to GPL.

However, I'd go for GPLv2+ at minimum or take the opportunity to switch to 3+ on this occasion.

guitorri, 2016-02-01:
ADMS shows up again as GPLv2+ in this lists of copyrights related to ngspice:
https://launchpad.net/ubuntu/xenial/+source/ngspice/+copyright

Actually the [ngspice/admst/ *](http://sourceforge.net/p/ngspice/ngspice/ci/master/tree/src/spicelib/devices/adms/admst/ngspiceMODULE.c.xml) contains XML scripts (not ADMS itself) written by the author of ADMS. The header of these file states this about the ADMS package:
>  `adms [...] is under [..] version 2 of the License, or (at your option) any later version.`

I just wanted  a confirmation that we can change the COPYING file to GPLv2+ as intended by the author.

It seems clear that ADMS is GPLv2+. It also seems to me that one cannot re-license it to GPLv2 (only), as it is taking rights away. Isn't is so?

@felix-salfelder I will look it up anyway, but if you can point out or comment (maybe a link to) the main differences between 2+ and 3+ I appreciate.

lutherthecat, 2016-02-01:
The instructions are clear in the COPYING file:

3. You may opt to apply the terms of the ordinary GNU General Public License instead of this License to a given copy of the Library. To do this, you must alter all the notices that refer to this License, so that they refer to the ordinary GNU General Public License, version 2, instead of to this License. (If a newer version than version 2 of the ordinary GNU General Public License has appeared, then you can specify that version instead if you wish.) Do not make any other change in these notices. 

Once this change is made in a given copy, it is irreversible for that copy, so the ordinary GNU General Public License applies to all subsequent copies and derivative works made from that copy. 

This option is useful when you wish to copy part of the code of the Library into a program that is not a library.

lutherthecat, 2016-02-01:
It doesn't matter what the intent of the author was.  You just need to go by the terms of the LGPL 2.1.

lutherthecat, 2016-02-01:
The conversion of version 2 to any later version is part of the copyright that is supposed to be put in the header file.  This option is not in the COPYING file.  Since the author of the program failed to put header files in admsXml, it is unclear whether they intended version 2 only.

felix-salfelder, 2016-02-03:
@guitorri https://www.gnu.org/licenses/rms-why-gplv3.en.html. "The reason to migrate is because of the existing problems that GPLv3 will address."

tl;dr; using GPLv2 instead of v3 only made sense when v3 did not exist yet.

felix-salfelder, 2016-02-03:
> put in the header file. [...]   failed to put header files

@lutherthecat not quite. from https://www.gnu.org/licenses/gpl-howto.html
"Whichever license you plan to use, the process involves adding two elements to each source file of your program: ..."

lutherthecat, 2016-02-03:
They didn't put it in any source file, which includes header files. 
Therefore the license was not clearly granted.

Regardless, the original copyright owner is the part of NXP that came 
from the part of Freescale, that came from the part of Motorola.  Please 
clarify the issue with them.

guitorri, 2016-02-06:
This copy of adms is now GPLv3+.

closed 2016-02-06 by guitorri