Qucs/ADMS #48: multiple constants

felix-salfelder, 2016-02-06:
there are duplicates (3) in constants.vams (my fault). anyway, there are 2 different electron charges.
this is curious. wikipedia says 1.60217646e-19 coulomb. where does the other stem from?

guitorri, 2016-02-06:
Ops. Did not see that one. Too bad I already pushed the tag and uploaded the tarball and binaries for 2.3.5... 

Anyway, there is another issue. It should catch the redefinition.

felix-salfelder, 2016-02-06:
don't worry. the release is fine. won't be the last `:D`

next time, maybe append -rc0 (-rc1..) to the tag/tarball version before the actual relese (unless you consider this overkill), so everyone can have a final look.

guitorri, 2016-02-06:
yep... I rushed things a bit this time... :smile:
I should have done a -rc or snapshot.

guitorri, 2016-10-28:
Fixed with #54

closed 2016-10-28 by guitorri