Qucs/ADMS #51: Windows binary: 64bit admsXml segfaults in msys64

guitorri, 2016-02-16:
The 64bit `admsXml` in the 64bit shell it crashes.

```
brondani@computer MINGW64 ~/git/qucs/qucs-core/src/components/verilog
$ admsXml.exe
[usage..] admsXml-2.3.5 (808929c) Feb  6 2016 20:17:31 (os=MSWIN64 compiler=MINGW32)
[usage..] admsXml source [options] -e script1 [-e script2 ...]

{snip}

$ gdb --args admsXml.exe EKV26MOS.va
GNU gdb (GDB) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from admsXml.exe...done.
(gdb) r
Starting program: C:\msys64\usr\local\bin\admsXml.exe EKV26MOS.va
[New Thread 8772.0x2980]
[info...] admsXml-2.3.5 (808929c) Feb  6 2016 20:17:31
[warning] [./EKV26MOS.va:18]: simplified vams header file created
[warning]   (not found in -I path) ... 'disciplines.vams'

Program received signal SIGSEGV, Segmentation fault.
0x000007feff781774 in msvcrt!fputs () from C:\Windows\system32\msvcrt.dll
(gdb) bt
#0  0x000007feff781774 in msvcrt!fputs () from C:\Windows\system32\msvcrt.dll
#1  0x00000000004144cc in adms_message_info_impl (
    format=0x478853 <yy_accept+1011> "s' header is available at\n",
    format@entry=0x478848 <yy_accept+1000> "standard '%s' header is available at\n") at adms.c:7053
#2  0x00000000004273da in adms_preprocessor_lex_include_file (
    fileName=<optimized out>) at preprocessorLex.l:148
#3  0x0000000000428b21 in preprocessorlex () at preprocessorLex.l:714
#4  0x000000000042b413 in preprocessorparse () at preprocessorYacc.c:1491
#5  0x00000000004044f4 in parseva (argc=2, argv=0x2d4ed0,
    myverilogamsfile=0x2d4fe0 "./EKV26MOS.va") at admsXml.c:622
#6  0x0000000000460a62 in main (argc=2, argv=0x2d4ed0) at admsXml.c:2776
(gdb) q
A debugging session is active.
```

With a 32bit `admsXml` version on the 64bit shell, it works:

````
brondani@computer MINGW64 ~/git/qucs/qucs-core/src/components/verilog
$ admsXml.exe -h
[usage..] admsXml-2.3.5 (808929c) Feb  6 2016 20:31:00 (os=MSWIN32 compiler=MINGW32)
[usage..] admsXml source [options] -e script1 [-e script2 ...]

{snip}

brondani@computer MINGW64 ~/git/qucs/qucs-core/src/components/verilog
$ admsXml.exe EKV26MOS.va
[info...] admsXml-2.3.5 (808929c) Feb  6 2016 20:31:00
[warning] [./EKV26MOS.va:18]: simplified vams header file created
[warning]   (not found in -I path) ... 'disciplines.vams'
[info...] standard 'disciplines.vams' header is available at
[info...]   'http://accellera.org/downloads/standards/v-ams'
[warning] [./EKV26MOS.va:19]: simplified vams header file created (not found in -I path) ... 'constants.vams'
[info...] standard 'constants.vams' header is available at 'http://accellera.org/downloads/standards/v-ams'
[warning] pragma redefined ... 'P_Q'
[warning] pragma redefined ... 'P_K'
[warning] pragma redefined ... 'P_CELSIUS0'
[info...] elapsed time: 0 (second)
[info...] admst iterations: 189786 (186678 freed)
```

guitorri, 2016-02-16:
Maybe escaping the single quotes in [here](https://github.com/Qucs/ADMS/blob/3e27d37d8ee81b8afd23d27f1a5f4e032d6033a5/admsXml/preprocessorLex.l#L148)

felix-salfelder, 2016-02-16:
is it #53? (sorry about that).

closed 2016-10-28 by guitorri