Qucs/ADMS #55: admsXml does not install and work for a new user under Ubuntu 14.04.3 LTS

RGD2, 2016-06-10:
Got the dependencies, then:
```
./configure
make
sudo make install
```

Appeared to work... but afterwards:
```
$ admsXml
admsXml: error while loading shared libraries: libadmsElement.so.0: cannot open shared object file: No such file or directory
```

I only need this because I was trying to install QUCS to try it, and QUCS unhelpfully just asks for it to be installed. (IMHO, it should just include it as a subtree if it needs it)

After this failed installation of ADMS, QUCS's ./configure fails with an error about not finding admsXml

I'm going to give up here and just use the PPA for QUCS 0.0.18, but it seems to me that the basic build-from-github flow shouldn't be this hard.

felix-salfelder, 2016-06-10:
> (IMHO, it should just include it as a subtree if it needs it)

does not make a lot of sense. qucs also uses qt, which is not included (to give one example). if you need qucs as part of something bigger, go ahead, create it.

but to the point: it seems, your system is slightly broken. and it wouldn't make much of a difference to install adms from a QUCS subtree (try).

please provide more details. can you confirm that

    $ which admsXml
    /usr/local/bin/admsXml

and

    $ ls  /usr/local/lib/libadmsElement.so.* -l
    lrwxrwxrwx [..] /usr/local/lib/libadmsElement.so.0 -> libadmsElement.so.0.0.0
    -rwxr-xr-x [..] /usr/local/lib/libadmsElement.so.0.0.0

? please provide the output of

    $ chrpath `which admsXml`

and of

    $ cat /etc/ld.so.conf /etc/ld.so.conf.d/*

> I'm going to give up here and just use the PPA for QUCS 0.0.18, but it seems to me that the basic build-from-github flow shouldn't be this hard.

the entanglement with admsXml is unfortunate. some have noticed, others don't care. nobody has time to fix it properly...

RGD2, 2016-06-15:
Apologies: I ended up just using the PPA, once I found it, so I no longer have my system in the same state. I need to press on with the job I wanted it for, and won't continue further at this time.

I think it's worthwhile to occasionally start with a fresh LTS Ubuntu installation, just to make sure the initial setup instructions for development-capable building still work. 
It also tends to make you reconsider them if they're particularly onerous, and this tends to help bring in fresh blood by lowering the barrier to entry a little bit.

I'll have another go when I have time - **but in the meantime if anyone can confirm they can install and build Qucs from github on a fresh Ubuntu install, I'll happily consider this a freak due to my particular config on this PC, and will close this.**

wrt to using a git subtree, my point was that unlike Qt, adms can't just be pulled in with in the `sudo apt-get`, and must come from a binary, or - ideally - be had without needing an invocation of `git submodule update` or friends. 

It looks like ADMS is orphaned otherwise, and it appears you've taken a fork anyway, so it may as well just be pulled in as a [git subtree](http://blogs.atlassian.com/2013/05/alternatives-to-git-submodule-git-subtree/), which in practise means one less step for new users as it would be pulled in with the initial git clone anyone does.

Anyway, merely a suggestion, so feel free to ignore it. :)
I can only apologise for not just fixing it myself - but I'd have to consult my notes merely to determine the actual number of things I have on my plate... I seem to recall more than a dozen at last count.

crobarcro, 2016-06-15:
You know, once upon a time we bundled ADMS sources with Qucs to make it easier to build. The reason for this was that ADMS was not in the repositories, and was pretty hard to get hold of. 

Has this situation changed? 

It doesn't seem like it, comparing it to Qt is a little spurious as Qt is on virtually every system out there. ADMS is not. A google search for 'Qt' also returns rather a lot of relevant results. A google search for 'ADMS' only turns up a link to a wikipedia article with no link to a place to get it from. You have to go searching to even figure out what it is.

Basically, it's a right pain to get hold of unless you're already 'in the know' which is why we bundled it with the Qucs sources in the first place. I say that until it's actually easy to get hold of we should put it back into the Qucs build system. Otherwise it puts up an annoying unnecessary barrier to entry to potential new contributors. The system we had before where the installed ADMS was used if found, but if not ADMS was built from source seemed like a reasonable solution until ADMS is more widely available.

guitorri, 2016-06-15:
License issues in ADMS should be resolved now. It could make it way into Debian. @bastien-roucaries was in contact to get it back into Debian. Other distros might have it already.

The distributed tarball from Sourceforge should be easy to build as `./configure && make install`. If that is not the case we should fix it. Building from this repository just requires a few more dependencies (Perl and its modules) but should be just as easy to build. 

@RGD2, without further info is difficult figure out what went wrong on your build.

felix-salfelder, 2016-06-15:
what we need much more urgent is to make adms optional on the qucsator and qucs gui end. if we do not pave the way for upcoming alternatives, then we will never see them.

in the meantime, everybody is permitted and welcome to bundle qucs with adms (and gcc and qt and whatnot. together with a kernel and on a bootable medium), whichever way. to my knowledge, nobody did (publicly). what does it mean?

if there's anything i can do about the debian package, please let me know...

in3otd, 2016-06-15:
>I'll have another go when I have time - **but in the meantime if anyone can confirm they can install and build Qucs from github on a fresh Ubuntu install, I'll happily consider this a freak due to my particular config on this PC, and will close this.**

@RGD2, FYI every proposed modification to the Qucs source is automatically compiled on Windows (using [AppVeyor](https://www.appveyor.com/)) and on OS X and Ubuntu (using [Travis](https://travis-ci.org/)) and only code that can be successfully compiled for all three platforms is accepted.This is not to say that this "build check" is perfect but we do try to make sure that the code can be compiled for some common environments.
Then, there are of course many possible variations and it can well be that some issues go unnoticed, so feedback from users is always welcome - and needed to improve the code!
I do sometimes check the build process on a particular Linux distribution in a VM, if a problem is reported - and when time permits - but for this a list of the exact steps that were performed is needed.

Travis builds Qucs with a bare Ubuntu 12.04 LTS (if I understood correctly), which release did you use?
But also it appears that it uses [`adms-2.3.4`](https://github.com/Qucs/qucs/blob/master/.travis.yml#L146) , while the latest version linked by the [README](https://github.com/Qucs/qucs/blob/master/README.md) is `adms-2.3.5`, is this correct? (@guitorri ?)

in3otd, 2016-06-15:
> what we need much more urgent is to make adms optional on the qucsator and qucs gui end. if we do not pave the way for upcoming alternatives, then we will never see them.

yes, making `adms` optional is the Right Thing, but probably it should have been pulled out of the Qucs sources only after this was done, as @crobarcro suggests.

> in the meantime, everybody is permitted and welcome to bundle qucs with adms (and gcc and qt and whatnot. together with a kernel and on a bootable medium), whichever way. to my knowledge, nobody did (publicly). what does it mean?

it simply means that those who know how to assemble such a bundle do not release it (why they should, they know how to install everything from sources and have already a working Qucs install - releasing and maintaining the bundle takes time) and those who have difficulties assembling everything from sources, silently drop Qucs and go elsewhere.

felix-salfelder, 2016-06-15:
yes and yes. priorities...

at the time, when i cleaned and fixed the build, i also pushed for a free (as in dfsg free) adms, which we now have (thanks to guilherme etc.). it was necessary to either drop or rewrite/repair the adms submodule thing, as it did not work right. eventually it was quite natural to drop it. better a hole in the row, than an aching tooth...

 rolling out a qucs+adms whichever way has become trivial (10 lines of shell script?). no, it's not on *my* list. but maybe somebody else is waiting for a qucs release?

> it simply means that those who know how to assemble such a bundle do not release it 

realistically, since adms has been dropped, nobody has released qucs either (the actual thing that takes time!). to make a long atory short: i once hoped that adms would be already optional long before the 0.0.19 release...

RGD2, 2016-06-15:
@in3otd : Cool, if Travis has got it, I'll consider this one closed. I had no problem with the prebuilt binaries from the PPA - maybe add a ref to it in the Readme on Qucs/qucs (under "Binary Installation")  in case new ppl come here first instead of reading the sf site? I did try ``sudo apt install qucs`` before googling qucs, and then went straight to the github repo - totally skipping the sf site and missing the point about the PPA. I'm probably weird in being overly-accustomed to building things from github, but I will use a PPA if I just need to run it.

This particular system has limited space on SSD, and it's a Ubuntu LTS with modifications from my Employer's IT department to conform with our security requirements, so it's fairly likely to have something odd broken. Probably their fault.

Thanks for the good work guys! 
QUCS has come a long way since I first met it, and it comfortably satisfies my needs in a circuit sim :)

closed 2016-06-15 by RGD2

cvgaviao, 2017-02-18:
Hi, I'm trying to install QUCS 0.19 in ubuntu and had to build ADMS from source as  qucs needs it as dependency.
I got the same original issue: 

```
cvgaviao@cvgaviao-Inspiron-5537:~$ ls /usr/local/lib/libadmsE* -l
-rw-r--r-- 1 root root 1493920 Fev 18 18:22 /usr/local/lib/libadmsElement.a
-rwxr-xr-x 1 root root     969 Fev 18 18:22 /usr/local/lib/libadmsElement.la
lrwxrwxrwx 1 root root      23 Fev 18 18:22 /usr/local/lib/libadmsElement.so -> libadmsElement.so.0.0.0
lrwxrwxrwx 1 root root      23 Fev 18 18:22 /usr/local/lib/libadmsElement.so.0 -> libadmsElement.so.0.0.0
-rwxr-xr-x 1 root root  645096 Fev 18 18:22 /usr/local/lib/libadmsElement.so.0.0.0
cvgaviao@cvgaviao-Inspiron-5537:~$ admsXml 
admsXml: error while loading shared libraries: libadmsElement.so.0: cannot open shared object file: No such file or directory
 
```
Please, what should I do ?

felix-salfelder, 2017-02-18:
just found. http://gnu-automake.7480.n7.nabble.com/Updating-the-shared-library-cache-automatically-using-ldconfig-td21094.html

i like the conclusion
> In the meantime, I think you could use the install-exec-hook target to
run ldconfig to update the cache.

@cvgaviao maybe run `ldconfig` (as root). it will update some library cache.

guitorri, 2017-02-18:
@felix-salfelder this means `install-exec-hook` might need to be patched, right?

Maybe `--disable-shared` can be used in the mean time to get an statically statically linked executable. So @cvgaviao can go on with his business of building QUCS.

felix-salfelder, 2017-02-18:
@guitorri: ideally, i would like to put

    install-exec-hook:
    \tif(we_are_root && we_don't_set_rpath && the_cache_is_outdated) then \
    \t  ldconfig
    \tfi

into `admsXml/Makefile.am`. (and not only there). i just don't have the time to spell it out, right now. would be nice (motivating....) to know if invoking `ldconfig` helped, wait and see

cvgaviao, 2017-02-19:
Hello @felix-salfelder and @guitorri, 
invoking **ldconfig** have worked! At least, I was able to invoke admsXML after install and it shows me the help instead of the error. thanks ! :)
But I got an error while compiling Qucs, but seems not to be related to adms. will open another issue.

cousteaulecommandant, 2020-07-03:
> maybe run `ldconfig` (as root). it will update some library cache.

Why is this not in the README?  Or at least a note saying that you might need to run that after installing if `admsXml -v` doesn't work.
(Btw I just ran into the same problem and this worked.  Ubuntu 18.04 in case that matters.)

felix-salfelder, 2020-07-03:
On Fri, Jul 03, 2020 at 09:49:57AM -0700, cousteau wrote:
> > maybe run `ldconfig` (as root). it will update some library cache.
> 
> Why is this not in the README?  Or at least a note saying that you
> might need to run that after installing if `admsXml -v` doesn't work.

since adms is in debian, this has become less interesting. but feel free
to send a patch.

note that this is not an adms issue, but an issue with your platform
configuration (specifically, ld.so.conf). there is a branch 'ldconfig',
which was meant to address this...

> (Btw I just ran into the same problem and this worked.  Ubuntu 18.04 in case that matters.)

thanks for your feedback.