Qucs/ADMS #56: Can`t install

AndriyVityk, 2016-07-14:
$ make
make  all-recursive
make[1]: вход в каталог «/home/andriy/software/adms-2.3.5»
Making all in scripts
make[2]: вход в каталог «/home/andriy/software/adms-2.3.5/scripts»
make[2]: Цель «all» не требует выполнения команд.
make[2]: выход из каталога «/home/andriy/software/adms-2.3.5/scripts»
Making all in admsXml
make[2]: вход в каталог «/home/andriy/software/adms-2.3.5/admsXml»
make[2]: *** Нет правила для сборки цели «verilogaYacc.y», требуемой для «verilogaYacc.c».  Останов.
make[2]: выход из каталога «/home/andriy/software/adms-2.3.5/admsXml»
Makefile:457: ошибка выполнения рецепта для цели «all-recursive»
make[1]: *** [all-recursive] Ошибка 1
make[1]: выход из каталога «/home/andriy/software/adms-2.3.5»
Makefile:389: ошибка выполнения рецепта для цели «all»
make: *** [all] Ошибка 2

felix-salfelder, 2016-07-15:
`$ LANG=C make`?

AndriyVityk, 2016-07-15:
Yes, I have not defined the compiler. Thank you, now it works.

closed 2016-10-28 by guitorri