Qucs/ADMS #57: Configuration Ok, No rule to make target `verilogaYacc.c'

JohnCC330, 2016-07-27:
I downloaded adms 2.3.5, and ran ./configure --prefix=/usr --libdir=/usr/lib64, with the following result:

```
adms version 2.3.5 configured successfully.

Configure Information:
  Host : x86_64-unknown-linux-gnu

  Maintainer mode : no

  C Compiler : gcc
  DEFS       :   -DHAVE_CONFIG_H
  CFLAGS     :   -g -O2  -Wall

  Linker     : /usr/x86_64-slackware-linux/bin/ld -m elf_x86_64
  LIBS       :   -lm 

  Prefix     : /usr
  Perl       : 
  Flex       : /usr/bin/flex
  Bison      : /usr/bin/bison

  Build shared library : yes

make  all-recursive
make[1]: Entering directory `/usr/local/src/electronics/ADMS-master'
Making all in scripts
make[2]: Entering directory `/usr/local/src/electronics/ADMS-master/scripts'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/usr/local/src/electronics/ADMS-master/scripts'
Making all in admsXml
make[2]: Entering directory `/usr/local/src/electronics/ADMS-master/admsXml'
make[2]: ***** No rule to make target `verilogaYacc.c', needed by `all'.  Stop.**
make[2]: Leaving directory `/usr/local/src/electronics/ADMS-master/admsXml'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/usr/local/src/electronics/ADMS-master'
make: *** [all] Error 2
Compilation (make) failed
```

Any hint as to the cause? (and solution, if possible)

System is Slackware 14.2, gcc (GCC) 4.9.3

felix-salfelder, 2016-07-28:
here it says

      Prefix     : /usr/local
      Perl       : /usr/bin/perl
      Flex       : /usr/bin/flex
      Bison      : /usr/bin/bison

so maybe this is a bug in configure (should fail if perl is missing?).

JohnCC330, 2016-07-28:
> should fail if perl is missing?
Perl is not missing. Though I don't use perl specifically, it's used in several scripts I have installed. Maybe the version? It's perl5 version 16.1

JohnCC330, 2016-07-28:
To be sure, I udated perl to 5.22 - no difference. And 'Perl:' in the configure result is still empty.

felix-salfelder, 2016-07-28:
had a look at configure.ac... it still uses "if ENABLE_MAINTAINER_MODE". uuh.

could you try `./configure --enable-maintainer-mode`? (isn't that what the README says?)

JohnCC330, 2016-07-28:
Hi Felix,

That seems to have solved the problem, thanks! 

Though I must say, this is the first time I _had_ to include --enable-maintainer-mode in configuring a project. I have compiled hundreds of autotools projects.

felix-salfelder, 2016-07-28:
> That seems to have solved the problem, thanks! 

not yet. but you are right that most projects do not make use of maintainer mode.

"Jim Meyering, the inventor of the AM_MAINTAINER_MODE macro was swayed by François’s arguments, and got rid of AM_MAINTAINER_MODE in all of his packages." (https://www.gnu.org/software/automake/manual/html_node/maintainer_002dmode.html)

please send us a patch, if you feel like it.

closed 2018-06-06 by guitorri