Qucs/ADMS #58: Support for rotational_omega discipline

mirandadam, 2016-10-20:
Hi, I would like to model a motor using ADMS and it seems ADMS/admsXml/disciplines.vams does not fully support the rotational_omega discipline with the basic  Omega and Theta accessors as in [http://verilogams.com/refman/basics/disciplines.html](url) and [http://accellera.org/downloads/standards/v-ams](url).

The disciplines.vams file has "FIXME" comments next to the rotational_omega definition. I figure one could use "V" instead of "Omega" and "I" (capital i) instead of "Theta" as things are right now.

* Is my above assumption about the accessors correct?
* Is this easy to fix? What needs to be done?
* How can I help?

felix-salfelder, 2016-10-22:
> How can I help?

please file a PR with the suggested change. it sounds reasonable.

mirandadam, 2016-10-22:
I don't have a suggested change yet... :-)
I will try just updating the disciplines.vams file and seeing what happens. If it works, I will submit a proper P.R. 

By the way, thanks for helping maintaining this great project!

felix-salfelder, 2020-08-18:
the discipline header included with adms is a minimal version, stripped down due to copyright. but there is no issue, as it is possible to use customized headers where needed. (still, feel free to send patches).

closed 2020-08-18 by felix-salfelder

mirandadam, 2020-08-27:
Sorry for leaving this open for so long. I have stopped working on the motor simulation long ago. Good thing you closed the issue.