Qucs/ADMS #62: Segmentation fault (core dumped)

HabelStefan, 2016-12-16:
Im trying to use adms with ubuntu.
When i try to convert a simple verilogA-Code

admsXml dummy.va -e adms.xml

, i get the error.

What works is
admsXml dummy.va -e dummy.xml

I tried a different model and once again 

admsXml resistor.va -e adms.xml does not work
but admsXml resistor.va -e dummy.xml does work.

Obviously the problem is the adms.xml.

I very new to all this and help would be very appreciated :)
Is there any tutorial source for adms?

Thank you.

ra3xdh, 2016-12-16:
@HabelStefan, Are you trying to use ADMSXML with Qucs? These presentations can help you. It explains interactions ADMS usage with Qucs.

http://www.mos-ak.org/london_2014/presentations/09_Mike_Brinson_MOS-AK_London_2014.pdf
http://www.mos-ak.org/berkeley_2014/presentations/04_Wladek_Grabinski_MOS-AK_Berkeley_2014.pdf
http://www.mos-ak.org/washington_dc_2015/presentations/T14_Mike_Brinson_MOS-AK_Washington_DC_2015.pdf

HabelStefan, 2016-12-16:
To be honest, i dont know at the moment as i am still trying to understand how it works ☺ I will check your sources.
Thank you very much!

http://www.mos-ak.org/london_2014/presentations/09_Mike_Brinson_MOS-AK_London_2014.pdf
http://www.mos-ak.org/berkeley_2014/presentations/04_Wladek_Grabinski_MOS-AK_Berkeley_2014.pdf
http://www.mos-ak.org/washington_dc_2015/presentations/T14_Mike_Brinson_MOS-AK_Washington_DC_2015.pdf

Wichtiger Hinweis: Die Information in dieser E-Mail ist vertraulich. Sie ist ausschließlich für den Adressaten bestimmt. Sollten Sie nicht der für diese E-Mail bestimmte Adressat sein, unterrichten Sie bitte den Absender und vernichten Sie diese Mail. Vielen Dank.
Unbeschadet der Korrespondenz per E-Mail, sind unsere Erklärungen ausschließlich final rechtsverbindlich, wenn sie in herkömmlicher Schriftform (mit eigenhändiger Unterschrift) oder durch Übermittlung eines solchen Schriftstücks per Telefax erfolgen.

Important note: The information included in this e-mail is confidential. It is solely intended for the recipient. If you are not the intended recipient of this e-mail please contact the sender and delete this message. Thank you. Without prejudice of e-mail correspondence, our statements are only legally binding when they are made in the conventional written form (with personal signature) or when such documents are sent by fax.

guitorri, 2016-12-23:
Which version of `admsXml` are you using.
Without your input files it might be difficult to reproduce.

Here you can find a copy of the original tutorials by the author of ADMS:
http://guitorri.github.io/adms.github.io/
https://github.com/guitorri/adms.github.io

The content was retrieved via [wayback machine](https://web.archive.org/web/20130202025457/http://vacomp.noovela.com/).
I cleaned up a bit and placed the input scripts as files ready for download.

felix-salfelder, 2020-09-10:
appears to be resolved (no feedback for a couple of years). closing.

closed 2020-09-10 by felix-salfelder