Qucs/ADMS #63: Error building with CMake from build/ directory

joe-skb7, 2017-02-06:
I'm trying to build the project like it described in README.md (section "Users Install from Tarball -> Compilation Using CMake"). But when executing this command (from `build/` directory)

    cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local

I encounter next errors:

Error 1:

    CMake Error at admsXml/CMakeLists.txt:176 (ADD_LIBRARY):
      Cannot find source file:

        /tmp/1/ADMS/build/admsXml/admstpathYacc.h

      Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
      .hxx .in .txx

Error 2:

    CMake Error at admsXml/CMakeLists.txt:109 (ADD_LIBRARY):
      Cannot find source file:

        /tmp/1/ADMS/build/admsXml/adms.c

      Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
      .hxx .in .txx

I managed to build the project running CMake from project's root directory, rather than from `build/`. The issue seems like hard-coded path in CMake file, which doesn't know that build path may differ from project's root directory.

guitorri, 2017-02-06:
Sorry for that. It has to do with the maintainer-mode thing. I replicated it on the CMake, but did not really test the out-of-tree build. I will see what I can do. Perhaps remove the maintainer-mode, which is the right thing to do anyway.

Infi9292, 2017-09-30:
I used the following command in Ubuntu and it has worked correctly.
cmake .. -DCMAKE_INSTALL_PREFIX=/home/me/local/adms -DUSE_MAINTAINER_MODE=ON

closed 2018-06-06 by guitorri