Qucs/ADMS #66: Print tree?

HabelStefan, 2017-05-17:
Is there a way to print the xml-tree of a parsed .va-file? Or is some kind of manual/documentation existing?

felix-salfelder, 2017-05-17:
there are a few example "scripts" in the scripts subdirectory. there is (afaik) no print-everyting script.

using the structure described in adms.xml (which i think is supposed to be self-documenting), it should be easy to write more scripts that write out more stuff in various ways.

be warned that using "XML scripts" for any kind of component model expansion might be the wrong track. please ignore this warning, if you are trying to debug/fix an existing implementation.

HabelStefan, 2017-06-06:
Thanks for the quick answer. I was on vacation therefore i took a while to answer.

What do i need in the .xml file to output the following module:

module myanalogfunction (p,q);
electrical p,q;
real a,b;

  analog   function real myfunction;
    input x;
    real  x;
    begin
      if(x<0.2)
        myfunction =  abs(x);
      else if(x<0.6)
        myfunction =  cos(2*x);
      else
        myfunction =  sin(2*x);
    end
  endfunction
endmodule

It should be seperated into

variables:
electrical
real
input

The function itself

Best regards!

felix-salfelder, 2017-06-06:
(no idea, would have to dig into that.)

i suggest to pick an existing template, find the place where it produces variable declarations. then strip off all the rest, try not to loose context.
then repeat with the other stuff you need.

felix-salfelder, 2020-08-18:
should be addressed in other issues by now.

closed 2020-08-18 by felix-salfelder