Qucs/ADMS #67: Analog Function "output" and "inout" variables do not get marked with dependencies

tvrusso, 2017-06-15:
I discovered (or rather, re-discovered) this issue while trying to process the [MVSG-HV HEMT model](https://nanohub.org/publications/73/1) for use in Xyce.  This model has a number of issues that complicated import into our simulator, but the hardest one to work around was this one.

ADMS correctly adds probe and other variable dependence to a variable's data in the data tree only if that variable appears on the left hand side of an assignment.   This covers almost all of the uses in most common Verilog-A models in the wild.

However, the MVSG-HV model makes use of some complicated analog functions that not only return a value for the function, but also have "output" and "inout" variables in their argument lists.  For all practical purposes, these amount to a potential assignment to the variables in those slots when the function is called.

ADMS will conservatively mark the variable on the left hand side of a call to such a function as depending on all the variables and probes in the argument list, but does NOT mark any of the output variables in this way.  This can lead to many problems, the most significant of which could be that elements of the jacobian data structure might never be created, if the only connection between variables occurs through dependencies introduced inside an analog function.  It also never marks "output" or "inout" variables as being assigned to, which complicates code generation (notably, if one is only declaring variables that get assigned to, as we do in Xyce's code generation templates).

I am attaching a tar file with some code that demonstrates this sort of problem.  The tar file contains two Verilog-A models that should implement identical models.  The first, "rlc.va" crudely implements a 2-terminal series RLC subcircuit in the most straightforward manner possible.  The second, "rlc_AF.va" moves all of the computation of currents, charges, and fluxes into an analog function that is very much like some of those in MVSG-HV, but is much, much simpler.

Also included are two HTML representations of data extracted from the ADMS data tree by the "html_params.xml" ADMST template file (also included in the tar file).  These HTML files can be viewed in any web browser.  The HTML files are N_DEV_ADMSrlc.html (corresponding to rlc.va) and N_DEV_ADMSrlc_AF.html (corresponding to rlc_AF.va).

Looking at N_DEV_ADMSrlc_AF.html shows that ADMS has not marked the "CapacitorCharge" and "InductorFlux" variables as depending on anything at all.   Correct variable dependencies are recorded for rlc.va, as shown by N_DEV_ADMSrlc.html.   Note also that due to the dependency issue, the rlc_AF.va module is missing three of the jacobian elements found for rlc.va, and rlc_AF.va has two additional jacobian elements not found in rlc.va (the extra elements are due to the issue described two paragraphs down, and are not really an error).

For rlc_AF.va, ResistorCurrent is only getting any of the dependencies marked because it appears on the LHS of the assignment as well as being an output variable.  

(Another difference here is that for rlc_AF.va, ADMS has marked ResistorCurrent as depending on *all* of the parameters and probes in the argument list, because it does this marking very conservatively, without a deep analysis of the analog function --- this is not a problem for us, but *can* produce an unnecessarliy dense jacobian stamp, as noted above).

I realize that fixing this may be a very difficult thing.  But ideally, any variable that appears as an argument to an analog function for an "inout" or "output" variable should be treated exactly as if it had appeared on the LHS of an assignment for the purposes of determining variable and probe dependence.

As an aside, the "html_params.xml" template is one I created for exploring the ADMS data tree prior to importing new models into Xyce, and may be of interest to others (c.f. issue #66).  To use it, run admsXml -e xyceBasicTemplates -e html_params.xml <verilogA input file>.  It will produce an HTML file with some useful tidbits about what appears in the parsed data tree.

[ADMS_AnalogFunctionIssue.tar.gz](https://github.com/Qucs/ADMS/files/1078699/ADMS_AnalogFunctionIssue.tar.gz)

The only workaround I've been able to find to get MVSG-HV to work for us is to hack on the verilog source and construct additional assignment lines to trick  ADMS into marking parameter and probe dependence, so that we can generate correct code from it.  It is an ugly workaround.

felix-salfelder, 2017-06-17:
thanks for the report -- it will be very useful to have more complete implicit rules...

I had a first look, see #68.

tvrusso, 2017-06-17:
Looks like a good start.  All of the dependencies (probe and variable) that would be transferred from RHS to LHS variables in an assignment should happen for output variables.  Probe dependencies are of course needed for normal simulation, but variable (especially parameter) dependencies could be important for sensitivity analysis.

tvrusso, 2019-08-08:
I have been working on this issue and believe I may have a fix that completely propagates all dependencies from analog function inputs to all outputs in exactly the same way that it already does to the return value.  I'm testing it out now, but so far it looks good.  Should have a PR to submit by this evening.

tvrusso, 2022-11-28:
Just a quick bump to note that I have added one more commit to the pull request associated with this issue.

The Xyce project is no longer using the adms.implicit.xml provided with ADMS and uses our own, and one of the things we have done is to fix this issue in our adms.implicitl.xml.  I had submitted PR#85 when I first coded it up in the hopes of sharing that work with the larger ADMS user base, but afterwards extended the fix somewhat in Xyce without updating the PR.

The PR now contains all of the analog-function-related fixes that Xyce uses and that wouldn't break existing back-ends.

Without these fixes, Verilog-A models that use analog functions with multiple output variables will not have the dependencies propagated properly.  Several CMC standard models (such as MVSG-HV (or as it is now called, MVSG-CMC)) make use of such analog functions, and code generators that make use of dependency information from the ADMS data tree will be getting incorrect information about these additional analog function outputs.