Qucs/ADMS #71: What to do with mapply_unary?

HabelStefan, 2017-06-26:
I'm still trying to figure out how to read the tree as the example script are, for me, not very useful to really understand how to get all the data from the tree.

One of the first problems that appears is that variables like var1=5.6 are saved as a number element while variables like var2=-3.5 are of type mapply_unary. The happens also when i use the example scripts that come with adms.

I never felt so lost :(

felix-salfelder, 2017-06-26:
try var1=+5.6. that should also get you a mapply_unary. note that no processing has been done. you get the structure as emitted by the parser. in these cases that is "a unary operator" applied to "the constant 5.6".

HabelStefan, 2017-06-26:
And how do i avoid it? I mean i need to use variables that are negative.

felix-salfelder, 2017-06-26:
what are you trying to achieve?

with adms you can essentially print out stuff in different ways. you may use some conditionals. e.g. "if it's a number, print that", if it's a "mapply_unary", print the sign, then print the argument (if it's a number). otherwise print an error...

HabelStefan, 2017-06-26:
First of all, thanks for helping me.
Basically i would like to read an verilog-file and printing it back in a file with some modifications. But before i can apply these modifications i need to understand how to print the original file.

My problem is that i yet did not understand the structure of the tree.

If there would be a way to print the whole tree :/.

felix-salfelder, 2017-06-26:
> If there would be a way to print the whole tree :/.

doesnt `%(.)` yield a string representantion of the full thing?

HabelStefan, 2017-06-26:
Obviously not or i doing it wrong.

<admst:for-each select="/">
<admst:message format="%(.)\n"/> 
</admst:for-each>
--> syntax error - invalid path "/"
or

<admst:message format="%(.)\n"/> 

--> no output

Is there a new tutorial beside the one for version 2.2.4?

tvrusso, 2017-06-26:
You should look at the script "vlatovla.xml" in the ADMS/scripts directory.  It will read a verilog module, process it, and write out an equivalent verilog file.  The script has all sorts of templates that demonstrate how things like mapply_unary should be handled.

You have to realize in something like 'variable=-5" that the thing on the RHS isn't a number as far as ADMS is concerned.  It's a unary operator (mapply_unary) taking a single argument.  In order to generate code from that, you have to handle mapply_unary, read its argument, and emit what you want to emit (in your case, exactly "-5").

Nothing is easy to do in ADMS.  

If you look at the vlatovla.xml script, you can begin to understand how you will have to structure your loops and what templates to apply just to emit exactly the same thing that you were inputting.

HabelStefan, 2017-06-28:
The "vlatovla.xml" helped me alot. Thank you!