Qucs/ADMS #73: Request clarification on "ddxprobe"?

tvrusso, 2017-07-11:
The "variable" element of the ADMS data tree has a "ddxprobe" sub-element.  Documentation says that this is represents that "the variable depends on these ddx probes."  Yet the ADMS implicit templates in adms.implicit.xml have this odd thing in the "modify" template around line 677:

      <admst:when test="[datatypename='assignment']">
       
        <admst:value-to test="$lhs/ddxprobe" select="#ddxprobe" string="yes"/>
        <admst:push into="rhs/variable/ddxprobe" select="$lhs/ddxprobe" onduplicate="ignore"/>
      </admst:when>

That is, the contents of the LHS's "ddxprobe" is copied into the RHS variables.  This is very odd, as, for example, this assignment:
            cjbot       =  TYPE * MULT_i * AB_i * ddx(qjunbot, V(A));
results in TYPE, MULT_i, AB_i, and qjunbot all having "V(A,GND)" in their ddxprobe elements, and cjbot NOT having it.  This is very confusing to me.  It suggests that TYPE, MULT_i, and AB_i need to be keeping around derivatives with respet to V(A,GND), despite the fact that nowhere in this module to these variables depend on V(A,GND) at all  (one of them is in fact an input parameter!).

It strikes me that perhaps this part of the implicit rules is just an error, and what was really intended was:
        <admst:push select="rhs/variable/ddxprobe" into="$lhs/ddxprobe" onduplicate="ignore"/>
i.e., to copy any existing ddxprobe information from RHS variables into the LHS.

It's not clear to me how ddxprobe is intended to be used.  Glancing at the Qucs ADMS file qucsVersion.xml (in qucs-core/src/components/verilog) it looks like it is intended to mean that the variable in question should have its derivatives with respect to the ddxprobe computed, but the odd propagation from left hand side to right makes it really unclear.

Can any of the Qucs/ADMS maintainers clarify the meaning of this data tree element, and how to use it properly?  I'm trying to add ddx() support to Xyce's ADMS code generation back end, and having a bear of a time making sense out of the data tree here.

felix-salfelder, 2017-07-11:
consider the assignments

    a = V(n)*c0 + c1;
    b = a;
    c = ddx(b, V(n))

my understanding is, that b is the first variable that "has a ddxprobe". then the ddxprobe is propagated from b to a (from lhs to rhs!), due to the second line. now "a has a ddxprobe", which will/should lead to code that computes the derivative of a wrt that ddxprobe.

i've played with this years ago, ended up with several unit tests for gnucap-adms. don't know how much of it ever worked, but it might give some ideas.

tvrusso, 2017-07-11:
Well.  That is about as clear as anything about ADMS.  So, because b is differentiated with respect to V(n) it has a ddx probe dependence, and so one must maintain a's derivatives with respect to V(n), too.  So the code snippet from qucsVersion.xml I saw is clearly not an error.

But in the case of the JUNCAP2000 model (from which I pulled the assignment in the initial question), I'm finding ADMS marking that input parameters such as TYPE have ddx probe dependence, when in fact they only appear as multiplicative factors *outside* the ddx.  So that is pretty strange.

But thank you for the clarification, it makes a lot more sense now.

tvrusso, 2022-01-22:
I am going to close this, as it was never properly an "issue" and was more appropriate for a discussion.  @felix-salfelder was kind enough to leave it open this long, but I should have closed it after getting his response.

closed 2022-01-22 by tvrusso