Qucs/ADMS #76: Lots of errors in bison-generated code

l29ah, 2018-02-18:
Trying to build develop branch on gentoo:
```
/bin/sh ../libtool  --tag=CC   --mode=compile x86_64-pc-linux-gnu-gcc -DHAVE_CONFIG_H -I. -I..  -DinsideadmstpathYacc   -O2 -pipe -O2 -pipe -march=nehalem  -Wall -c -o libadmsAdmstpath_la-admstpathYacc.lo `test -f 'admstpathYacc.c' || echo './'`admstpathYacc.c
libtool: compile:  x86_64-pc-linux-gnu-gcc -DHAVE_CONFIG_H -I. -I.. -DinsideadmstpathYacc -O2 -pipe -O2 -pipe -march=nehalem -Wall -c admstpathYacc.c  -fPIC -DPIC -o .libs/libadmsAdmstpath_la-admstpathYacc.o
admstpathYacc.y: In function ‘location030sr1’:
admstpathYacc.y:1364:9: error: ‘t_admst {aka struct s_admst}’ has no member named ‘_ps’
     e=ai->_ps==NULL) p->_a1=p->_an=d; else p->_an->_nexta=d,d->_preva=p->_an,p->_an=d;d->_position=++p->_position; return;
         ^
admstpathYacc.y:1364:20: error: expected ‘;’ before ‘)’ token
     e=ai->_ps==NULL) p->_a1=p->_an=d; else p->_an->_nexta=d,d->_preva=p->_an,p->_an=d;d->_position=++p->_position; return;
                    ^
admstpathYacc.y:1364:20: error: expected statement before ‘)’ token
admstpathYacc.y:1364:36: error: ‘d’ undeclared (first use in this function)
     e=ai->_ps==NULL) p->_a1=p->_an=d; else p->_an->_nexta=d,d->_preva=p->_an,p->_an=d;d->_position=++p->_position; return;
                                    ^
admstpathYacc.y:1364:36: note: each undeclared identifier is reported only once for each function it appears in
admstpathYacc.y:1364:39: error: ‘else’ without a previous ‘if’
     e=ai->_ps==NULL) p->_a1=p->_an=d; else p->_an->_nexta=d,d->_preva=p->_an,p->_an=d;d->_position=++p->_position; return;
                                       ^
admstpathYacc.y:1364:60: warning: left-hand operand of comma expression has no effect [-Wunused-value]
     e=ai->_ps==NULL) p->_a1=p->_an=d; else p->_an->_nexta=d,d->_preva=p->_an,p->_an=d;d->_position=++p->_position; return;
                                                            ^
admstpathYacc.y:1364:77: warning: left-hand operand of comma expression has no effect [-Wunused-value]
     e=ai->_ps==NULL) p->_a1=p->_an=d; else p->_an->_nexta=d,d->_preva=p->_an,p->_an=d;d->_position=++p->_position; return;
                                                                             ^
admstpathYacc.y:1362:9: warning: variable ‘e’ set but not used [-Wunused-but-set-variable]
   admse e;
         ^
admstpathYacc.y: At top level:
admstpathYacc.y:1366:3: error: expected identifier or ‘(’ before ‘if’
   if(0) {}
   ^
admstpathYacc.y:1367:3: error: expected identifier or ‘(’ before ‘if’
   if(0) {}
   ^
admstpathYacc.y:1368:3: error: expected identifier or ‘(’ before ‘else’
   else if(e==admse_variable)
   ^
admstpathYacc.y:1373:3: error: expected identifier or ‘(’ before ‘else’
   else if(e==admse_variableprototype)
   ^
admstpathYacc.y:1377:3: error: expected identifier or ‘(’ before ‘else’
   else
   ^
admstpathYacc.y:1382:1: error: expected identifier or ‘(’ before ‘}’ token
 }
 ^
admstpathYacc.y:1383:13: error: redefinition of ‘location030input’
 static void location030input(p_ptraverse p,p_path mypath,p_admst dot)
             ^
admstpathYacc.y:1297:13: note: previous definition of ‘location030input’ was here
 static void location030input(p_ptraverse p,p_path mypath,p_admst dot)
             ^
admstpathYacc.y:1410:13: error: redefinition of ‘location030direction’
 static void location030direction(p_ptraverse p,p_path mypath,p_admst dot)
             ^
admstpathYacc.y:916:13: note: previous definition of ‘location030direction’ was here
 static void location030direction(p_ptraverse p,p_path mypath,p_admst dot)
             ^
…
```

```
‰ yacc --version
bison (GNU Bison) 3.0.4
Written by Robert Corbett and Richard Stallman.

Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

in3otd, 2018-02-18:
I'm not familiar with how the adms build works but I see that those yacc files were autogenerated by a perl script. I suspect something went wrong in the previous steps, can you post a full build log ?

l29ah, 2018-02-18:
On Sun, Feb 18, 2018 at 08:13:00AM +0000, in3otd wrote:
> I'm not familiar with how the adms build works but I see that those yacc files were autogenerated by a perl script. I suspect something went wrong in the previous steps, can you post a full build log ?

https://dump.bitcheese.net/files/pyfesys/fileF3UycT

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

flip111, 2019-02-14:
I have the same problem. Can not use qucs on ubuntu 16.04 because of this.

felix-salfelder, 2020-08-18:
seems to relate to an old version and/or an old yacc version. please try with recent tools and/or reopen with a patch.

closed 2020-08-18 by felix-salfelder