Qucs/ADMS #82: implicit.xml node: case not handled for bsimbulk.va

promach, 2019-01-23:
I am having problem using [bsimbulk.va](https://bsim.berkeley.edu/models/bsimbulk/)

The [admsXML instruction](http://qucs.sourceforge.net/docs/report/verilog.pdf#page=3) gives me the following error:

>     phung@UbuntuHW15:~/BSIM-BULK_106.2.0_20170630/code$ admsXml bsimbulk.va -e /usr/local/include/qucsator/qucsVersion.xml -e /usr/local/include/qucsator/qucsMODULEcore.xml
>     [info...] admsXml-2.3.6 (e24d007) Jan 22 2019 14:34:12
>     [fatal..] node: case not handled
>     [fatal..] see [.adms.implicit.xml:314]
>     phung@UbuntuHW15:~/BSIM-BULK_106.2.0_20170630/code$

When I look at the [Qucs verilog-A application note](https://github.com/Qucs/ADMS/files/2785556/Qucs_Verilog_ANotes.pdf) , it also runs the similar admsXML commands.

Could anyone help with the error ?

Note: https://github.com/Qucs/qucs/pull/595 does not really solve the issue since for the compilation of va source file to c++, we still need admsXML backend in the terminal window as described in section V in page 3 of the [Qucs verilog-A application note](https://github.com/Qucs/ADMS/files/2785556/Qucs_Verilog_ANotes.pdf). Please correct me if wrong.

Please also see this [sourceforge comment about the temperature node](https://sourceforge.net/p/qucs/discussion/311050/thread/16111ffa/#b0b5). I have quoted the comment below as well.

> I believe the issue is related with the temperature node. It does not know how to handle it.
> If I comment out all the code involving t and (t) things gets translated to C++. I did not try to compile and run.

@felix-salfelder @guitorri @la3pna @in3otd

in3otd, 2019-01-23:
maybe the [modifications](https://sourceforge.net/p/ngspice/ngspice/ci/bbf4047f409c59d7b6829876c774dfb5c52425b2/) done in `ngspice` to make it work with `admsXml` could be helpful

tvrusso, 2019-01-24:
This is almost certainly related to the BSIM group's use of "port_connected", which they use to make the node optional --- ADMS tends to hate it.  The function is not recognized, and the fact that it has a node name as an argument appears to be what generates the cryptic "node: case not handled" error.

We have the same problem with other BSIM models in Xyce's ADMS back-end, and generally have to hack the verilog to take out all uses of port_connected and work around the optional node issue somehow (generally by manually post-processing the C++ we output).

tvrusso, 2019-01-24:
FWIW, the predecessor of bsimbulk, bsim6, also had use of port_connected, but inside an ifdef that was default off (__THERMAL_NODE__).  They appear to have removed the ifdef and made the thermal node default here.

For other BSIM models (e.g. BSIM-CMG) that don't have it ifdefed, the Xyce team was unable to use the verilog unmodified, and had to hack on it by making the thermal node unconditionally internal (e.g. by removing the use of $port_connected)  just to get it to process properly in ADMS, then hack on the C++ output further to turn the thermal node back into an optional one the "xyce way".

felix-salfelder, 2019-01-27:
On Sun, Jan 27, 2019 at 03:29:28AM -0800, promach wrote:
> @guitorri Would you be able to share your modified bsimbulk.va (turned
> off temperature code segment) here ?

FWIW: I suggested to create a qucs-nonfree repo for this kind of stuff
some time ago... (it exists now, https://github.com/Qucs/qucs-nonfree).

promach, 2019-02-16:
@tvrusso I have the following fatal error using [BSIM6 6.1.1 model](http://bsim.berkeley.edu/BSIMBULK/BSIM6.1.1_07302015.tar.gz)

> [phung@archlinux code]$ admsXml BSIM6.1.1.va -e /usr/local/include/qucsator/qucsVersion.xml -e /usr/include/qucsator/qucsMODULEcore.xml
> [info...] admsXml-2.3.6 (e24d007) Dec 13 2016 22:57:39
> [fatal..] option -e: cannot read file '/usr/local/include/qucsator/qucsVersion.xml'
> [phung@archlinux code]$ admsXml BSIM6.1.1.va -e /usr/include/qucsator/qucsVersion.xml -e /usr/include/qucsator/qucsMODULEcore.xml
> [info...] admsXml-2.3.6 (e24d007) Dec 13 2016 22:57:39
> [warning] bsim6: device not handled by the adms qucs interface
> [warning] please ensure extra code to be added to the interface
> [info...] bsim6.core.cpp and bsim6.core.h: files created
> [info...] elapsed time: 8 (second)
> [info...] admst iterations: 26513433 (25190536 freed)
> [phung@archlinux code]$ admsXml BSIM6.1.1.va -e /usr/include/qucsator/qucsVersion.xml -e /usr/include/qucsator/qucsMODULEdefs.xml
> [info...] admsXml-2.3.6 (e24d007) Dec 13 2016 22:57:39
> [warning] bsim6: device not handled by the adms qucs interface
> [warning] please ensure extra code to be added to the interface
> [info...] bsim6.defs.h: file created
> [info...] elapsed time: 1 (second)
> [info...] admst iterations: 5264800 (5243119 freed)
> [phung@archlinux code]$ admsXml BSIM6.1.1.va -e /usr/include/qucsator/qucsVersion.xml -e /usr/include/qucsator/qucsMODULEgui.xml
> [info...] admsXml-2.3.6 (e24d007) Dec 13 2016 22:57:39
> [warning] bsim6: device not handled by the adms qucs interface
> [warning] please ensure extra code to be added to the interface
> [fatal..] Template not found:
> [fatal..]   <admst:template match=":mapply_ternary"/>
> [fatal..]   see:  [/usr/include/qucsator/qucsMODULEgui.xml:22]
> [phung@archlinux code]$

tvrusso, 2019-02-16:
I'm sorry  --- while I have experience using ADMS for Xyce (a completely different project than Qucs, but which has its own ADMS back-end), and have gotten BSIM6 working in Xyce, I have no specific knowledge of how Qucs's ADMS code generation back-end works, so I can't help you.  Sounds like there's something missing from the qucsMODULEgui.xml templates, but that's as much as I can tell.   You'll have to get help from Qucs developers on that one.

promach, 2019-02-28:
This is a bit out of context for qucs adms, but I am not sure.

What do you guys think about Xyce (using qucs adms backend) compilation process ? I have faced some issue [during qucs adms importing for Xyce](https://groups.google.com/forum/#!topic/xyce-users/v5tofQLaybI)

tvrusso, 2019-02-28:
You're right.  Talking about Xyce's compilation process is way off topic for this issue.  It would be better discussed on the Xyce project's google group.

closed 2020-08-18 by felix-salfelder