Qucs/ADMS #89: New release?

tvrusso, 2020-07-28:
ADMS has seen a number of useful changes in its develop branch since its last release in 2016 (release 2.3.6).  This is fine for those of us that live off of the bleeding edge and compile from source out of github, but means that these changes are unavailable to those who depend on release versions through Linux package managers --- few of which build from anything but releases.

It would probably be useful to merge the current develop branch to master and release it as 2.3.7, even if there is no enthusiasm for merging some of the open PRs.

felix-salfelder, 2020-07-29:
On Tue, Jul 28, 2020 at 10:08:32AM -0700, Tom Russo wrote:
> [..] current develop branch to master and release it as 2.3.7 [..]

yes. Perhaps for a release we still need to update NEWS and ChangeLog.

felix-salfelder, 2020-08-18:
done

closed 2020-08-18 by felix-salfelder