Qucs/ADMS #90: Does CentOS 7 support admsXML?

HSV2PVG, 2020-08-20:
On CentOS 6, it was supported and when I switched from 6 to 7, following the instruction to install admsXML based on http://ngspice.sourceforge.net/admshowto.html. System told me to 
configure: error: Perl package XML::LibXML may be downloaded from http://search.cpan.org/dist/libXML

I tried all the the installation of the related files, but still can't get it to work. Any suggestions?

felix-salfelder, 2020-08-31:
Have you tried Adms from here (i.e. Qucs/ADMS)?

> sudo apt-get install libxml2-dev zlib1g-dev

Sounds a bit like Debian. Adms is available here. try `apt install adms`.

HSV2PVG, 2020-09-01:
> Have you tried Adms from here (i.e. Qucs/ADMS)?
> 
> > sudo apt-get install libxml2-dev zlib1g-dev
> 
> Sounds a bit like Debian. Adms is available here. try `apt install adms`.

I tried and still didn't work

HSV2PVG, 2020-09-01:
Ok, I updated my gcc and it is 6.1.0 and bison version is 3.0.4. 
now I had the same issue as this post: https://sourceforge.net/p/qucs/mailman/message/31835159/
I followed to this place: https://bugs.freedesktop.org/show_bug.cgi?id=67354
but how to do that next? 

here is what I did
>./autogen_lin.sh
>make

 y.tab.c: In function 'admstpathparse':
> y.tab.c:14933:7: error: too few arguments to function 'admstpathlex'
> admstpathYacc.y:8:12: note: declared here
>   static int admstpathlex (p_pparse mypparse);
>              ^
> admstpathYacc.y:13643:43: error: 'mypparse' undeclared (first use in
> this function)
>      p_text mytext=adms_text_new(((p_pparse)mypparse)->_transform,"");
>                                             ^
> admstpathYacc.y:13643:43: note: each undeclared identifier is reported
> only once for each function it appears in
> admstpathYacc.y: In function 'pparse':
> admstpathYacc.y:14618:5: error: too many arguments to function 'admstpathparse'
>       admstpathparse (mypparse);
>       ^
> admstpathYacc.c:63:25: note: declared here
>   #define yyparse         admstpathparse
>                           ^
> y.tab.c:14777:1: note: in expansion of macro 'yyparse'
> admstpathYacc.y: In function 'tparse':
> admstpathYacc.y:14651:5: error: too many arguments to function 'admstpathparse'
>       admstpathparse (mypparse);
>       ^
> admstpathYacc.c:63:25: note: declared here
>   #define yyparse         admstpathparse
>                           ^
> y.tab.c:14777:1: note: in expansion of macro 'yyparse'
> admstpathYacc.y: In function 'apath_main':
> admstpathYacc.y:14690:5: error: too many arguments to function 'admstpathparse'
>       admstpathparse (mypparse);
>       ^
> admstpathYacc.c:63:25: note: declared here
>   #define yyparse         admstpathparse
>                           ^
> y.tab.c:14777:1: note: in expansion of macro 'yyparse'

felix-salfelder, 2020-09-01:
On Tue, Sep 01, 2020 at 12:43:16AM -0700, HSV2PVG wrote:
> Ok, I updated my gcc and it is 6.1.0 and bison version is 3.0.4.
> now I had the same issue as this post: https://sourceforge.net/p/qucs/mailman/message/31835159/
> I followed to this place: [...]

I can not see autogen_lin.sh. are you sure this is the right place?

you could try https://github.com/Qucs/ADMS/releases/tag/release-2.3.7.
basically, you do, `configure`, followed by `make`, see INSTALL.

HSV2PVG, 2020-09-01:
I followed the step on their website: http://ngspice.sourceforge.net/admshowto.html
> Under OS LINUX (tested with SUSE 11.2, 64 bit) you may expand the zip file and run ./autogen_lin.sh, followed by 'make' and 'make install'.
Let me tried yours again. Thanks!

felix-salfelder, 2020-09-10:
please reopen if this is not resolved.