Qucs/ADMS #98: Cano not parse BSIM-CMG 111.0.0 model

lehaianh1986, 2021-12-25:
Hi,

I have tried newest ADMS tool to parse BSIM-CMG 111.0.0 model from http://bsim.berkeley.edu/models/bsimcmg/ but it can not understand some new syntax such as `length_scaling

Can you support this newest model into ADMS?

tvrusso, 2021-12-26:
The problem is not that we don't support "`length_scaling" (this is just a macro definition defined in bsimcmg_macros.include), but that there is some sort of syntax problem in the nested ternary operators used in the "lexp" macro, which is in turn used by the length_scaling macro.

When parentheses are added to the lexp macro do make its use of ternaries more easily parsed, other problems in the model are revealed which may require significant work to get past.  The first one I encountered is the use of a variable in a collapse conditional that ADMS is declaring as a local variable but which needs to be scoped as "global_instance."  Dealing with that sort of thing in BSIM models requires modifying the Verilog-A and cannot be addressed in the Xyce/ADMS back-end.

The BSIM-CMG 110 model currently in Xyce required some heavy handed hackage to get working in Xyce/ADMS.  It was not trivial, and those changes are documented in the file "utils/ADMS/examples/BSIMCMG110.0.0_20160101/code/README_Xyce.  In fact, the Xyce team has not yet found any Verilog-A of a BSIM model that compiled cleanly in ADMS without significant work.

I would not expect the 111 model to be any easier,  and it would be unrealistic to expect it to compile with Xyce/ADMS right out of the box.

tvrusso, 2021-12-26:
Looking over the BSIM-CMG 111 source code suggests to me that getting it to work in Xyce will be even harder than the 110 model was, because they have removed some of the code structure that made it easier to port to Xyce.

A trend these days is to claim that that Verilog-A authors should never have to worry about isolating bias-independent quantities for optimization, because a good compiler can just optimize them for you.

That presumes a good optimizing Verilog-A compiler, which ADMS is not.  Therefore, any time an author has taken this guidance to heart and *removed* structures that clearly define bias-independent quantities they have made it that much harder to work in Xyce (where the first step is to identify where those quantities are computed and re-insert code hints that ADMS needs to identify them and generate correct code).

At some point the Xyce team may have to spend time getting the BSIM CMG 111 into Xyce, at which time we'll probably have a lot of work to do in the CMG 111's Verilog-A.  This is not a matter that can simply be tweaked in Xyce/ADMS to fix parsing.  it's a collection of little problems in the model itself that make it incompatible with the limited range of ADMS.

But the Xyce team is unlikely to have adoption of the BSIM-CMG 111 model into Xyce as a high priority any time soon.

tvrusso, 2021-12-26:
I apologize for my comments (which I have deleted).  I did not pay attention to which project this issue was made to, and thought it was to Xyce's issue tracker, not ADMS's.  My comments were entirely based on the assumption that this was a Xyce/ADMS question, not an ADMS question.

My point about the problem being with the structure of the "lexp" macro usage in the "length_scaling" macro is still valid, though --- there is some kind of error in the use of nested ternary operators that can be fixed with a few parentheses.

lehaianh1986, 2021-12-27:
@tvrusso thanks for your information. I'm still looking if there are some expert with ADMS who have experimented on this model.

yiz98, 2023-02-15:
> @tvrusso thanks for your information. I'm still looking if there are some expert with ADMS who have experimented on this model.

Hi. I'm having the same problem. I'm wondering if the solution is found. Thanks.

lehaianh1986, 2023-02-16:
> > @tvrusso thanks for your information. I'm still looking if there are some expert with ADMS who have experimented on this model.
> 
> Hi. I'm having the same problem. I'm wondering if the solution is found. Thanks.

No, it is still unresolved.

ra3xdh, 2023-02-16:
@lehaianh1986 You may try the OpenVAF compiler https://openvaf.semimod.de/ It is actively developed as ADMS replacement. You may submit this issue to OpenVAF team if necessary.