Qucs/ADMS #113: `elsif directive supported?

dwarning, 2023-09-20:
LRM 2.4.1 allows:
`ifdef THIS
this code
`elsif THAT
that code
`else
other code
`endif

With my admsXml version 2.3.7 I have the impression THAT code is never reached - falling through to other code.
In which file the directives are defined? Can I extend them?

tvrusso, 2023-09-20:
A very quick scan of ADMS's preprocessor lexer and parser shows that it supports only ifdef, ifndef, else, and endif, not elsif.

Odd that it doesn't puke on the elsif even though its preprocessor lexer and parser don't recognize it.

I did a quick test by creating:
  foobar.va:
```
`ifdef THIS
  foo
`elsif THAT
  bar
`else
  baz
`endif
```

and running it through admsXml:
> admsXml -DTHIS=1 -e ~/src/ADMS/scripts/vlatovla.xml foobar.va 
[info...] admsXml-2.3.7 (f05b7d6ac7d6) Sep 15 2021 09:30:43
[info...] [./foobar.va:3]: macro `elsif is undefined
[fatal..] ./foobar.va: during lexical analysis syntax error at line 2 -- see 'foo'
> less .foobar.va.adms 
# 1 "./foobar.va"

  foo
 THAT
  bar
# 8 "./foobar.va"

So in fact it *does* emit an error on the elsif, just not a fatal one.  And then essentially it does the wrong thing and passes both the if and elsif stuff through with a junk line (stripping elsif and leaving just the condition).

dwarning, 2023-09-20:
May be to simple, but can we extend
preprocessorLex.l
preprocessorYacc.y

Should be similar to

`R_ifdef
        : TK_PRAGMA_NAME
          {
            $$=$1;
            if(!DONT_SKIPP)
            {
              adms_slist_push(&pproot()->skipp_text,INT2ADMS(1));
              adms_slist_push(&condistrue,INT2ADMS(-1));
            }
            else if(adms_preprocessor_identifier_is_def($1))
            {
              adms_slist_push(&condistrue,INT2ADMS(1));
              adms_slist_push(&pproot()->skipp_text,INT2ADMS(0));
            }
            else
            {
              adms_slist_push(&condistrue,INT2ADMS(0));
              adms_slist_push(&pproot()->skipp_text,INT2ADMS(1));
            }
          }
        ;
`

tvrusso, 2023-09-20:
Pretty sure there would be more to it than that.  Need to modify the R_conditional section and provide means for handling multiple elsif blocks.

I looked at it briefly in the hope that it would be simple enough to cram in and make a PR, but it's more than I'm willing to work on.